Atlas-Scientific/Ezo_I2c_lib #3: Error Compliling

TNTechnologies, 2019-07-22:
I am getting a call of overload is ambiguous error.

/home/kb001/sketchbook/libraries/Ezo_I2c_lib/Ezo_i2c.cpp: In member function ‘void Ezo_board::send_cmd_with_num(const char*, float, uint8_t)’:
/home/kb001/sketchbook/libraries/Ezo_I2c_lib/Ezo_i2c.cpp:33:56: error: call of overloaded ‘String(float&, uint8_t&)’ is ambiguous
  String temp = String(cmd )+ String(num, decimal_amount);
                                                        ^
In file included from /usr/share/arduino/hardware/arduino/cores/arduino/Arduino.h:192:0,
                 from /home/kb001/sketchbook/libraries/Ezo_I2c_lib/Ezo_i2c.h:28,
                 from /home/kb001/sketchbook/libraries/Ezo_I2c_lib/Ezo_i2c.cpp:2:
/usr/share/arduino/hardware/arduino/cores/arduino/WString.h:70:11: note: candidate: String::String(long unsigned int, unsigned char)
  explicit String(unsigned long, unsigned char base=10);
           ^
/usr/share/arduino/hardware/arduino/cores/arduino/WString.h:69:11: note: candidate: String::String(long int, unsigned char)
  explicit String(long, unsigned char base=10);
           ^
/usr/share/arduino/hardware/arduino/cores/arduino/WString.h:68:11: note: candidate: String::String(unsigned int, unsigned char)
  explicit String(unsigned int, unsigned char base=10);
           ^
/usr/share/arduino/hardware/arduino/cores/arduino/WString.h:67:11: note: candidate: String::String(int, unsigned char)
  explicit String(int, unsigned char base=10);
           ^
/usr/share/arduino/hardware/arduino/cores/arduino/WString.h:66:11: note: candidate: String::String(unsigned char, unsigned char)
  explicit String(unsigned char, unsigned char base=10);
           ^

Atlas-Scientific, 2019-07-22:
Do the examples compile? Which version of the IDE are you using?

TNTechnologies, 2019-07-22:
IDE Version 2:1.0.5+dfsg2-4.1
OS: 5.1.0-parrot1-3t-amd64 #1 SMP Parrot 5.1.3-1parrot1.3t (2019-05-20) x86_64 GNU/Linuxe

The examples don't compile either. For now I do not need to send any number to the EC so i have commented out the function in the .cpp file and was able to compile it fine and is running well on the unit as we speak.

Atlas-Scientific, 2019-07-22:
So according to my research that's a really old version of the IDE (https://forum.arduino.cc/index.php?topic=370405.0). Please upgrade to a recent version and see if the problem persists.

TNTechnologies, 2019-07-22:
That Solved it. I did not realize it was not being maintained by my package manager. Thank you for the help. Great Products!

closed 2019-07-22 by TNTechnologies

julianreistq, 2022-05-25:
I have the same error in a newer version of the Arduino IDE.
Is there another possible fix to this problem?