Atlas-Scientific/Ezo_I2c_lib #10: Calibrating doesn't return Ok

mzamot, 2021-02-25:
Running any calibration command is not returning Ok status, but the sensor name and pin number only:

> ph:cal,mid,7
  -PH: 99

Despite this, the calibration works.

Atlas-Scientific, 2021-02-25:
The I2C protocol doesn't return *ok, it returns a response code followed by a message. If the response code is success with no message, that's what the result looks like. Otherwise it would either give an error, or append the message. That said, we will think about ways of making the success confirmation more obvious.

smjacques, 2023-09-03:
Considering this, how to calibrate the other pH solutions for the curve? Because in my case, it doesn't update the serial monitor, plus the pH values seem to be outside the range of the solutions I'm measuring. I've also tried the examples iot_cmd_sample_code.ino and legacy_hydroponics_kit_with_DO.ino with the same problem.

smjacques, 2023-10-06:
Yes, I'm following that with a 3 points calibration.
The problem is that when I enter the serial command for calibration, I can't see the data output and when calibration values stabilize for each standard. The final verification shows me that calibration seems okay (?cal, 3), although the correct would be to have the values during the calibration to be sure it is reliable.

[image]

Atlas-Scientific, 2023-10-06:
You have several options: you can use the "poll" command to restart the polling after every calibration, or you can send it "R" to take individual readings from the sensor you're calibrating.  The code stops polling when you give it commands because that makes it less messy, but the poll command turns the polling back on.

closed 2025-01-07 by Atlas-Scientific