Atlas-Scientific/Ezo_I2c_lib #17: Error Compiling

julianreistq, 2022-05-25:
I have the same error in a newer version of the Arduino IDE.
Is there another possible fix to this problem?

_Originally posted by @julianreistq in https://github.com/Atlas-Scientific/Ezo_I2c_lib/issues/3#issuecomment-1136896194_

julianreistq, 2022-05-25:
`void Ezo_board::send_cmd_with_num(const char* cmd, float num, uint8_t decimal_amount){
	String temp = String(cmd )+ String(num);  //, decimal_amount); 
	const char* pointer = temp.c_str();
	send_cmd(pointer);
}`
Removing the decimal_amount from the String(num) fixed it for me.

bbombek, 2022-05-28:
it doesnt compile what to do??
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None, Default"

C:\Users\Boris The Jack\Documents\Arduino\libraries\Ezo_I2c_lib-master\Ezo_i2c.cpp: In member function 'void Ezo_board::send_cmd_with_num(const char*, float, uint8_t)':

C:\Users\Boris The Jack\Documents\Arduino\libraries\Ezo_I2c_lib-master\Ezo_i2c.cpp:41:56: error: call of overloaded 'String(float&, uint8_t&)' is ambiguous

  String temp = String(cmd )+ String(num, decimal_amount);

                                                        ^

In file included from C:\Users\Boris The Jack\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\cores\esp32/Arduino.h:159,

                 from C:\Users\Boris The Jack\Documents\Arduino\libraries\Ezo_I2c_lib-master\Ezo_i2c.h:28,

                 from C:\Users\Boris The Jack\Documents\Arduino\libraries\Ezo_I2c_lib-master\Ezo_i2c.cpp:2:

C:\Users\Boris The Jack\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\cores\esp32/WString.h:75:18: note: candidate: 'String::String(double, unsigned int)'

         explicit String(double, unsigned int decimalPlaces = 2);

                  ^~~~~~

C:\Users\Boris The Jack\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\cores\esp32/WString.h:74:18: note: candidate: 'String::String(float, unsigned int)'

         explicit String(float, unsigned int decimalPlaces = 2);

                  ^~~~~~

C:\Users\Boris The Jack\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\cores\esp32/WString.h:73:18: note: candidate: 'String::String(long unsigned int, unsigned char)'

         explicit String(unsigned long, unsigned char base = 10);

                  ^~~~~~

C:\Users\Boris The Jack\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\cores\esp32/WString.h:72:18: note: candidate: 'String::String(long int, unsigned char)'

         explicit String(long, unsigned char base = 10);

                  ^~~~~~

C:\Users\Boris The Jack\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\cores\esp32/WString.h:71:18: note: candidate: 'String::String(unsigned int, unsigned char)'

         explicit String(unsigned int, unsigned char base = 10);

                  ^~~~~~

C:\Users\Boris The Jack\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\cores\esp32/WString.h:70:18: note: candidate: 'String::String(int, unsigned char)'

         explicit String(int, unsigned char base = 10);

                  ^~~~~~

C:\Users\Boris The Jack\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\cores\esp32/WString.h:69:18: note: candidate: 'String::String(unsigned char, unsigned char)'

         explicit String(unsigned char, unsigned char base = 10);

                  ^~~~~~

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\Boris The Jack\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi

exit status 1

Error compiling for board Adafruit ESP32 Feather.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bbombek, 2022-05-28:
Hello!
I did as is on the atlas tutorial for the new WiFi pool kit and i get the massage like this.
[image]

Please help me!
Thank you 
Boris

bbombek, 2022-06-01:
Guys at atlas helped me in a day
do this and your problems are away!!
Thank you atlas scientific for prompt answer!!!

[@bbombek](https://github.com/bbombek) As a quick fix try to adjust a line of code in class Ezo_board::send_cmd_with_num inside the Ezo_i2c.cpp file.
The line:
String temp = String(cmd )+ String(num, decimal_amount); to
String temp = String(cmd )+ String(num)

Atlas-Scientific, 2022-06-01:
So it took some effort to replicate this issue and find the cause, and it seems like its been caused by espressif's [latest update](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3) somehow. Specifically [this bug fix](https://github.com/espressif/arduino-esp32/pull/6138). Its strange because the API allows the decimal amount to be set, there shouldn't be an issue there. I'm working on a more permanent fix, but if not I'll do something like the fix suggested here despite it breaking the decimal setting functionality. If any of you can find a way to make it work without breaking said functionality, let me know.

Another fix to this btw is to downgrade the esp32 library to 2.0.2 using the board manager in the Arduino IDE. See step 4B in [this document](https://files.atlas-scientific.com/Wi-Fi-Hydroponics-kit-setup-guide.pdf) on how to get to the board manager

bbombek, 2022-06-01:
this did the trick
THANK YOU
Boris
You are the best

Atlas-Scientific, 2022-06-01:
Alright I believe [this commit](https://github.com/Atlas-Scientific/Ezo_I2c_lib/commit/9462286131f1ece9faf41a8527d0b5ae337b1124) should fix it. Let me know if it doesn't work for you.

closed 2022-06-01 by Atlas-Scientific

bbombek, 2022-10-11:
another problem
Now it works but i want to flash with a new code and (let say I put a new
pump and change some code)
now it is not in a bootloader mode??
How do I reset the ESP32 into bootloader?
So I can put new cote on via arduino IDE?

[image: image.png]

Sketch uses 681565 bytes (51%) of program storage space. Maximum is 1310720
bytes.
Global variables use 38824 bytes (11%) of dynamic memory, leaving 288856
bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.3
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode
detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit:
https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
the selected serial port For troubleshooting steps visit:
https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
 does not exist or your board is not connected