Atlas-Scientific/Ezo_I2c_lib #19: Problem with OLED display library

rodri16, 2022-07-20:
I run  I2c_read_mulitple_circuits.ino with no issues but when adding an OLED display on the same I2C bus readings are 0.
The library I use [U8g2](https://github.com/olikraus/u8g2)

I am using TTGO LoRa32 v2.1.6 Board. [ttgo lora32 v2.1.6 Board ](http://www.lilygo.cn/prod_view.aspx?TypeId=50003&Id=1130&FId=t3:50003:3)

This is the code, with _#define OLED_ enable or disable the OLED.

```
#define OLED
#include <Ezo_i2c.h> //include the EZO I2C library from https://github.com/Atlas-Scientific/Ezo_I2c_lib
#include <Wire.h>    //include arduinos i2c library
#include <sequencer2.h> //imports a 2 function sequencer 
#include <Ezo_i2c_util.h> //brings in common print statements

Ezo_board PH = Ezo_board(99, "PH");       //create a PH circuit object, who's address is 99 and name is "PH"
Ezo_board EC = Ezo_board(100, "EC");      //create an EC circuit object who's address is 100 and name is "EC"
Ezo_board RTD = Ezo_board(102, "RTD");      //create an EC circuit object who's address is 100 and name is "EC"
Ezo_board DO = Ezo_board(97, "DO");
Ezo_board ORP = Ezo_board(98, "ORP");

void step1();  //forward declarations of functions to use them in the sequencer before defining them
void step2();

Sequencer2 Seq(&step1, 1000, &step2, 1000);  //calls the steps in sequence with time in between them

/* OLED display setup*************************************************/
#ifdef OLED
#include <U8x8lib.h>  //https://github.com/olikraus/u8g2
U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/*rst*/ U8X8_PIN_NONE);
#endif

void setup() {
#ifdef OLED
  u8x8.begin();
  // reduce I2C bus speed (must be done after begin)
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  u8x8.drawString(0, 0, "Start");
  delay(1000);
  u8x8.setPowerSave(1); //LCD off
#endif
  Wire.begin(); //start the I2C
  Wire.setClock(100000L);//400000L
  Serial.begin(115200);                     //start the serial communication to the computer

  Seq.reset();                            //initialize the sequencer
}

void loop() {
  Seq.run();                              //run the sequncer to do the polling
}

void step1() {
  //send a read command. we use this command instead of PH.send_cmd("R");
  //to let the library know to parse the reading
   Wire.begin(); //start the I2C
   PH.send_read_cmd();
  EC.send_read_cmd();
  RTD.send_read_cmd();
  DO.send_read_cmd();
  ORP.send_read_cmd();
}

void step2() {
  receive_and_print_reading(RTD);             //get the reading from the PH circuit
  Serial.print("  ");
  receive_and_print_reading(DO);             //get the reading from the PH circuit
  Serial.print("  ");
  receive_and_print_reading(ORP);             //get the reading from the PH circuit
  Serial.print("  ");
  receive_and_print_reading(PH);             //get the reading from the PH circuit
  Serial.print("  ");
  receive_and_print_reading(EC);             //get the reading from the EC circuit
  Serial.println();

  EC.send_cmd("Sleep");
  DO.send_cmd("Sleep");
  PH.send_cmd("Sleep");
  ORP.send_cmd("Sleep");
  RTD.send_cmd("Sleep");
  
#ifdef OLED
  float pH1 = PH.get_last_received_reading();
  float EC1 = EC.get_last_received_reading();
  float DO1 = DO.get_last_received_reading();
  float ORP1 = ORP.get_last_received_reading();
  float Temp = RTD.get_last_received_reading();

  u8x8.setPowerSave(0); //LCD off
  String stra;
  stra = "pH:" + String(pH1, 1);
  u8x8.drawString(0, 1, stra.c_str());
  stra = "T:" + String(Temp); //Temp
  u8x8.drawString(0, 2, stra.c_str());
  stra = "DO:" + String(DO1);
  u8x8.drawString(0, 3, stra.c_str());
  stra = "ORP:" + String(ORP1);
  u8x8.drawString(0, 4, stra.c_str());
  stra = "EC:" + String(EC1);
  u8x8.drawString(0, 5, stra.c_str());
  delay(1000);
  u8x8.setPowerSave(1); //LCD off
#endif
}
```

Atlas-Scientific, 2022-07-20:
How exactly are the ezo circuits responding? Do you see the lights blink as if they're taking readings, or do they blink red (indicating errors), or do they not respond at all? What kind of response is printed by the receive_and_print_reading function?

rodri16, 2022-07-21:
No different response, light blinking are the same (green then blue then off) but I read null values. Response printed is 0
This behavior is not shown on RTC which always works Ok.

Atlas-Scientific, 2022-07-21:
Can you show me what your serial monitor prints out when running this code?

rodri16, 2022-07-22:
I have plugged only EC and RTD sensors, DO board is not present. Take a look to EC value
With OLED lib
RTD: 26.64  DO: No Data   ORP: 233.10  PH: 0.00  EC: 0.00  
Without OLED lib
 RTD: 26.70  DO: No Data   ORP: 595.50  PH: 0.00  EC: 8263.00

Atlas-Scientific, 2022-07-22:
So its only EC that reads 0 and not all the sensors? Its possible that the screen and EC have the same I2C address. If so try changing the address of the EC, and make sure it doesn't conflict with any of your other I2C addresses.

rodri16, 2022-07-25:
No, it's not that problem. With a I2C scanner I detect all devices with different addresses. OLED address is 60

I2C scanner. Scanning ...
Found address: 60 (0x3C)
Found address: 98 (0x62)
Found address: 99 (0x63)
Found address: 100 (0x64)
Found address: 102 (0x66)

rodri16, 2022-07-25:
I realized that first measure is right but then it gives zero
EC: 8918.00  RTD: 28.25  ORP: 1020.00  PH: 0.00  
EC: 0.00  RTD: 28.25  ORP: 1020.00  PH: 0.00 
Between first and second display is on and showing data

If I run the code without OLED no problem at all
EC: 8930.00  RTD: 28.28  ORP: 1020.00  PH: 0.00  
EC: 8997.00  RTD: 28.28  ORP: 1020.00  PH: 0.00  
EC: 9005.00  RTD: 28.28  ORP: 1020.00  PH: 0.00

rodri16, 2022-07-25:
PROBLEM SOLVED!!!
It seems to be that OLED library changes I2C speed, so adding Wire.setClock(100000L); before reading solves the problem!
Thanks

closed 2022-07-25 by rodri16