Atlas-Scientific/Ezo_I2c_lib #20: Allow providing custom I2C bus

gaaf, 2023-03-23:
Hi,

I need to use an other I2C bus than the default, but the library is hardcoded to use ```Wire()```. Please provide a constructor or method to set the I2C bus (TwoWire object) to use.

Something like this: https://github.com/RobTillaart/PCF8575/blob/701bdbe40d7204a3a263a0b2fb99cf06802c9291/PCF8575.h#L31

moritz89, 2023-03-23:
I created a fork for this reason :sweat_smile: 

- https://github.com/InamataCo/ezo-i2c-sensors/blob/master/Ezo_i2c.h
- https://registry.platformio.org/libraries/mulmer89/EZO%20I2C%20Sensors

Might be of help...

gaaf, 2023-03-23:
Thanks. I already adapted it myself. It looks eeringly similar though...

Atlas-Scientific, 2023-03-29:
If you already wrote it and it's simple, can you just do a pull request with the changes? Or are there some issues with your implementation?

moritz89, 2023-03-31:
Created an MR here: #21

Atlas-Scientific, 2023-04-03:
Alright I added it with some tweaks, feel free to test it and let me know if it works for you

moritz89, 2023-04-03:
I'm not using the sensors ATM, but it should work. I had tested the code on my fork, but a confirmation would be great!

Atlas-Scientific, 2023-04-03:
I tested it with hardware to some extent so that's fine, I just mean if you or anyone else needs other things in this API

closed 2025-01-07 by Atlas-Scientific