Atlas-Scientific/Ezo_I2c_lib #32: i2c.py turning off i2c ports

clock-matt, 2025-05-28:
So I've been trying to set up my sensors to communicate with my raspberry pi using this i3 interlink board. But when I run i2c.py it seems like it's turning off the sensors? Below is my error log from it. 

The only way I've found thus fat to turn them back on is to reboot the pi. The AtlasIOTCore is finding them just fine though. Same resuts happen whether I have AtlasIOTCore running or not, as far as I can tell. I will update this post with more information as I continue troubleshooting. 

run detect on boot

i2cdetect -y 1

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- 2c -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 4f
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- 61 -- -- -- -- 66 -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

now run program

sudo python i2c.py

>> Atlas Scientific I2C sample code
>> Any commands entered are passed to the default target device via I2C except:
  - Help
      brings up this menu
  - List
      lists the available I2C circuits.
      the --> indicates the target device that will receive individual commands
  - xxx:[command]
      sends the command to the device at I2C address xxx
      and sets future communications to that address
      Ex: "102:status" will send the command status to address 102
  - all:[command]
      sends the command to all devices
  - Poll[,x.xx]
      command continuously polls all devices
      the optional argument [,x.xx] lets you set a polling time
      where x.xx is greater than the minimum 1.50 second timeout.
      by default it will poll every 1.50 seconds
>> Pressing ctrl-c will stop the polling

Traceback (most recent call last):
  File "/home/pioreactor/Raspberry-Pi-sample-code/i2c.py", line 150, in <module>
    main()
  File "/home/pioreactor/Raspberry-Pi-sample-code/i2c.py", line 65, in main
    device_list = get_devices()
                  ^^^^^^^^^^^^^
  File "/home/pioreactor/Raspberry-Pi-sample-code/i2c.py", line 28, in get_devices
    response = device.query("I")
               ^^^^^^^^^^^^^^^^^
  File "/home/pioreactor/Raspberry-Pi-sample-code/AtlasI2C.py", line 169, in query
    return self.read()
           ^^^^^^^^^^^
  File "/home/pioreactor/Raspberry-Pi-sample-code/AtlasI2C.py", line 135, in read
    raw_data = self.file_read.read(num_of_bytes)

now if i run detect again

i2cdetect -y 1

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

clock-matt, 2025-06-05:
Additionally, I have connected the same sensors on a separate pi and i3 interlink board, but for that set up I get a different error instead. On that set up I get:

>> Atlas Scientific I2C sample code
>> Any commands entered are passed to the default target device via I2C except:
  - Help
      brings up this menu
  - List 
      lists the available I2C circuits.
      the --> indicates the target device that will receive individual commands
  - xxx:[command]
      sends the command to the device at I2C address xxx 
      and sets future communications to that address
      Ex: "102:status" will send the command status to address 102
  - all:[command]
      sends the command to all devices
  - Poll[,x.xx]
      command continuously polls all devices
      the optional argument [,x.xx] lets you set a polling time
      where x.xx is greater than the minimum 1.50 second timeout.
      by default it will poll every 1.50 seconds
>> Pressing ctrl-c will stop the polling
    
>> WARNING: device at I2C address 97 has not been identified as an EZO device, and will not be queried
>> WARNING: device at I2C address 102 has not been identified as an EZO device, and will not be queried
Traceback (most recent call last):
  File "/home/pioreactor/Raspberry-Pi-sample-code/i2c.py", line 150, in <module>
    main()
  File "/home/pioreactor/Raspberry-Pi-sample-code/i2c.py", line 67, in main
    device = device_list[0]
             ~~~~~~~~~~~^^^
IndexError: list index out of range

Atlas-Scientific, 2025-06-05:
So you must have AtlasIOTCore turned off and ensure its not running if you want to use other software to communicate with the Atlas devices. Even if you close the browser it still runs in the background. If you run both software at the same time they will interfere with each other.

clock-matt, 2025-06-05:
The set up that throws the first error, that turns off the i2c ports, throws the same error whether I have AtlasIOTCore running or not. I will try running i2c.py on the other set up without AtlasIOTCore running (I currently having it run on start up).

Atlas-Scientific, 2025-06-05:
What do the LEDs on the devices do? And which Pi are you using?

clock-matt, 2025-06-05:
Attaching the HAT to the second set up did replicate the error from the first set up. I'm going to try modifying the polling now.

clock-matt, 2025-06-05:
I excluded the range of i2c channels being scanned in AtlasI2C.py and it fixed the error! Thank you so much for the help.

Atlas-Scientific, 2025-06-05:
Alright, awesome